armbian/linux-rockchip #8: sync with khadas edge2 kernel

amazingfate, 2023-03-13:
diff from khadas and rockchip's sdk: https://github.com/khadas/linux/compare/487de35b82fd72b78058b7c5efa34e9a59e0bb87...984d3c8991003cc6920b457eb430a0ba4742f0ca

amazingfate, 2023-03-13:
There are two commits which are no need to get merged:
https://github.com/khadas/linux/commit/1596f502a313b8ff7a9dcf9bb4632915e6170a50
https://github.com/khadas/linux/commit/1eaa8ef55d1df998ac8543441a6105e4833e0efc

amazingfate, 2023-03-13:
I use `git format-patch -k 487de35b82fd72b78058b7c5efa34e9a59e0bb87..984d3c8991003cc6920b457eb430a0ba4742f0ca -o patches` to generate patches from commits. We only need deveicetree and drivers related commit, and there are many commits with both driver and devicetree, we have to split them.

efectn, 2023-05-17:
I guess this can be closed now

closed 2023-05-18 by amazingfate