armbian/linux-rockchip #113: lcd_vk2c21 - driver that can't be disabled

ilyakurdyukov, 2023-10-29:
I'm trying to make the config that would have the same drivers as the Android kernel from the device.

And found a driver that I can't disable: `/sys/bus/platform/drivers/lcd_vk2c21`

[drivers/char/Makefile](https://github.com/armbian/linux-rockchip/blob/0d05f8c90fbdcf35e63fa58c0d0400a19bc3a33f/drivers/char/Makefile#L50C27-L50C28):

```makefile
#fox.luo@add vk2c21 lcd driver
obj-y	+= lcd_vk2c21.o
```

Who's accepted this? Such low-quality drivers shouldn't be allowed.

amazingfate, 2023-10-29:
Pull request is welcomed.

ilyakurdyukov, 2023-10-29:
@rpardini, please improve the vendor hacks you brought so that this driver can be disabled properly and only compiles for blueberry edge and not for every Armbian kernel for Rockchip.

amazingfate, 2023-10-29:
> @rpardini, please improve the vendor hacks you brought so that this driver can be disabled properly and only compiles for blueberry edge and not for every Armbian kernel for Rockchip.

All legacy rk3588 boards share the same kernel config, so this is impossible.

ilyakurdyukov, 2023-10-29:
> All legacy rk3588 boards share the same kernel config, so this is impossible.

There should be proper config switch here, `CONFIG_LCD_VK2C21` for example, and `CONFIG_LCD_VK2C21=y` added to the RK3588 config. I don't want to fix it myself because I won't be able to verify that the changes are correct.

```patch
 obj-$(CONFIG_XILLYBUS)		+= xillybus/
 obj-$(CONFIG_POWERNV_OP_PANEL)	+= powernv-op-panel.o
 obj-$(CONFIG_ADI)		+= adi.o
-#fox.luo@add vk2c21 lcd driver
-obj-y	+= lcd_vk2c21.o
+obj-$(CONFIG_LCD_VK2C21)	+= lcd_vk2c21.o
```

Or such hack patches should go to the armbian-build as standalone patches, and not be committed to linux-rockchip.
This is okay for the vendor, and not for the repository, where the kernel is common to different devices.

amazingfate, 2023-10-29:
This is a common kernel for all rockchip devices, so it's good to get all drivers compiled. This driver can get built as module so it will get loaded depend on devicetree.

ilyakurdyukov, 2023-10-29:
> This is a common kernel for all rockchip devices, so it's good to get all drivers compiled. This driver can get built as module so it will get loaded depend on devicetree.

In its current state, this driver cannot be built as a module because it is not defined properly. This is added as a hack and always compiles into the kernel, which is what I'm upset about.

amazingfate, 2023-10-29:
I'm okay with the current state. And pull request is welcomed.

rpardini, 2023-10-29:
I agree. This driver was sent this way by the vendor. It should indeed have proper Kconfig and Makefile conditional. PR's welcome.

ilyakurdyukov, 2023-10-29:
If no one fixes it first, I'll fix it when I figure out my current problem with the RK3528. It is better for this to be fixed by some of the owners of the device that needs this driver.

> This driver was sent this way by the vendor.

By the way, do you have instructions and templates for sending letters to a vendor so that they open the source code of the missing drivers?

rpardini, 2023-10-29:
No. Contact the EFF / FSF / etc

closed 2023-10-29 by rpardini