armbian/linux-rockchip #132: patch/kernel/media-current/00184-linux-1001-v4l2-rockchip.patch fails for station-m2, current, edge OK

rossbcan, 2023-12-20:
BOARD: station-m2
BRANCH: Current

git status:
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   config/kernel/linux-media-current.config
	modified:   config/kernel/linux-media-edge.config
	modified:   config/kernel/linux-meson64-current.config
	modified:   config/kernel/linux-meson64-edge.config

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	boot.cmd
	build.sh
	interfaces
	meson-gxm-tx9-pro-100M.dtb
	meson-gxm-tx9-pro.dtb
	readme.txt
	rk3566_bootfs_files/
	rk3566_rootfs_files/
	s912_bootfs_files/
	s912_rootfs_files/
``
config/kernel/linux-media-current.config changes: minor, select OpenVfd, disabled patch to add OpenVfd, so is defconfig
The untracked files are for post-processing after compile
SO: BSP is virgin up to date.
Selecting station-m2, current and compiling will fail as reported. Edge branch compiles OK.

Build Log: command failed: {"message":"Error adding document."} Server down?

[image]

My last issue was indeed my fault, so being careful, insuring minimal / no changes to build environment

Regards;
Bill Ross

igorpecovnik, 2023-12-20:
Official build framework bug tracker is here https://github.com/armbian/build/issues and it was not possible to reproduce. You don't have latest framework or you use some fork.

rossbcan, 2023-12-20:
so...
damn, sorry to bother you
"git status:
On branch main
Your branch is up to date with 'origin/main'."

how do I determine if pull necessary?

Thanks;
Bill

closed 2023-12-20 by rossbcan

igorpecovnik, 2023-12-20:
Git handling ... stash your changes away to not loose them, ...

```
armbian:build:% git reset --hard HEAD^
HEAD is now at 363a06902 Disable u-boot generation also for EDGE branch
armbian:build:% git pull
```

and check if you have this (ATM):

git log --name-status HEAD^..HEAD
```
commit a82589e683471911cad04f4d4615bb6a3656eb85 (HEAD -> main, origin/main, origin/HEAD)
Author: Gunjan Gupta <viraniac@gmail.com>
Date:   Wed Dec 20 17:03:38 2023 +0530

```
then add your changes and compile.