armbian/linux-rockchip #154: rebase from 5.10 to rockchip 6.1

amazingfate, 2024-03-03:
@Joshua-Riek has done some rebase: https://github.com/Joshua-Riek/linux-rockchip/tree/rk-6.1-rkr1
I also did a rebase without building.
Here is the list of dropped patches
| code module | commit name | drop reason |
| ---- | ---- | --- |
| net | https://github.com/armbian/linux-rockchip/commit/81c2a4ce71e4e0e912fb3d17cfdf570098844f5e |  |
| net | https://github.com/armbian/linux-rockchip/commit/4699501901a5e0325fe789c4d8e4098cdc66dac6 |  |
| net | https://github.com/armbian/linux-rockchip/commit/a86585fbea3b84b71b87d4a655d2c5293aeafd56 |  |
| net | https://github.com/armbian/linux-rockchip/commit/d425ee128eb94e1d455a2ad2d4481f0871f91a21 |  |
| net | https://github.com/armbian/linux-rockchip/commit/be5f28943310dec69d090a34342797f5b61f7af9 |  |
| net | https://github.com/armbian/linux-rockchip/commit/5df52303a36fdbbb2a19a4cdbb8e7d89f31c3dae |  |
| net | https://github.com/armbian/linux-rockchip/commit/c7596ba2cee145a9be953bede76f4e71f8ebdaf1 |  |
| net | https://github.com/armbian/linux-rockchip/commit/69ac97c70c5e9447c900b13b529421b3e8e8aa60 |  |
| net | https://github.com/armbian/linux-rockchip/commit/1211e705126f7ddf1c1713d5aa0534f5ac6cd0ae |  |
| net | https://github.com/armbian/linux-rockchip/commit/dc87545ce727e0fdd4e15e9111a55d38b778c126 |  |
| net | https://github.com/armbian/linux-rockchip/commit/e05a708576e861af4bcea0e6d06fc7b2bf534530 |  |
| net | https://github.com/armbian/linux-rockchip/commit/0d05f8c90fbdcf35e63fa58c0d0400a19bc3a33f |  |
| khadas | https://github.com/armbian/linux-rockchip/commit/c608f287ef21fe0dbc7e0390cc45c21bf9f73b6b |  |
| khadas | https://github.com/armbian/linux-rockchip/commit/2b1cb9d3f6120333cb24a5143e26de1168a6284c |  |
| khadas | https://github.com/armbian/linux-rockchip/commit/af7c34605fda3c6ed03869765cfa4fb84e90b73b |  |
| build | https://github.com/armbian/linux-rockchip/commit/7c622d095d168a706444bd4e88937a5b324ec550 |  |
| build | https://github.com/armbian/linux-rockchip/commit/f0fb25570af93e12cb8e832c859dd7c963018075 |  |
| build | https://github.com/armbian/linux-rockchip/commit/3e2980446878c6bb4f41a0207687d8e4d0ecb41b |  |
| build | https://github.com/armbian/linux-rockchip/commit/94d4102a30cdd4c99a391fd481c15ea23860b2cb |  |
| dma-buf | https://github.com/armbian/linux-rockchip/commit/4e4095c4ed4ad88feed39e851c797cbe6f7a9a2c |  |
| dma-buf | https://github.com/armbian/linux-rockchip/commit/edb2457f29572efb0d5ef8331701dd4f6ccab937 |  |
| dma-buf | https://github.com/armbian/linux-rockchip/commit/afbe8bc95e38dd5c90ef44bba8764ade4a882e2f |  |
| display | https://github.com/armbian/linux-rockchip/commit/1e0050d30d7d2bd3f7cc952b9b184d9ff0adabfd |  |

Joshua-Riek, 2024-03-03:
Hey! For panfork and wayland to work you need the below commits:

https://github.com/Joshua-Riek/linux-rockchip/commit/6ee34db111dddc085bb7f063367aa242f1f196f0
https://github.com/Joshua-Riek/linux-rockchip/commit/32fc58106ab6e03f60b062423b8148bc43ce030d
https://github.com/Joshua-Riek/linux-rockchip/commit/9e1e2c5cc83eff4339b34fb600e88a39563aba93

Btw do you have your 6.1 branch pushed?

amazingfate, 2024-03-04:
My rebased code has not passed building. I will push it after it's built successfully and tested.

amazingfate, 2024-03-10:
closed as pr is created: https://github.com/armbian/linux-rockchip/pull/155

closed 2024-03-10 by amazingfate