armbian/linux-rockchip #164: rk-6.1-rkr1: captured video from IMX415 at MIPI/CSI not connected to ISP block, on rk3588s platform

enok71, 2024-04-10:
On my rock 5a I can capture video from my IMX415 camera connected to the 4 lane CSI port if I capture data from /dev/video11. But trying to get data from /dev/video[01] (mainpath/selfpath) doesn't work. The reason seems to be related to the fact that device node /dev/v4l-subdev4 (rkcif-mipi-lvds2) is not created at boot. There seems to be no connection from rkcif to rkisp. All this worked in 5.10 but doesn't work in 6.1. I suspect it is an issue for all rk3588s platforms.

See also discussion and details [here](https://forum.radxa.com/t/dev-video-01)

Revision tested: [27f6561551948be6350c9bbd549c84acb2e559af](https://github.com/armbian/linux-rockchip/commit/27f6561551948be6350c9bbd549c84acb2e559af)

enok71, 2024-04-15:
I registered this issue at radxa kernel github https://github.com/radxa/kernel/issues/298

amazingfate, 2024-07-04:
You may try new rkr3 kernel to see if rockchip has fixed your issue: https://github.com/armbian/linux-rockchip/actions/runs/9740746614

belveder79, 2025-03-21:
Just to confirm that this is still an issue with rk-6.1-rkr5 and NanoPC-T6 - different camera, but same issue about /dev/v4l-subdev4

diverger, 2025-06-13:
Yes, I've struggled on this several days, it seems the code has dependency dead lock in the data to isp flow.

tizbac, 2025-07-26:
> For me the IMX415 does not work *at all* on the Radxa 5b+ (the official
> Radxa 4k camera).
> Using latest Armbian with vendor kernel.
> I does work with the official Radxa image (that i can't use in production
> for other reasons).
> Attached is the dmesg log. Let me know if I need to provide any additional
> logs.
> dmesg-imx-fail.txt

I built armbian image with legacy kernel ( 5.10-rkr8 and for me it works,
so it is something that is failing to load on 6.1

belveder79, 2025-09-27:
BUMP! 
This is still broken rk-6.1-rkr5....

levonti, 2025-09-28:
> I built armbian image with legacy kernel ( 5.10-rkr8 and for me it works,
> so it is something that is failing to load on 6.1
> 
> Il ven 25 lug 2025, 22:35 TheOnlyJoey ***@***.***> ha scritto:
> […](#)

Could you explain how you build armbian with legacy kernel? is no build legacy core in the menu for board rock-5b-plus, for example. 

P.S. I added legacy item to the menu by analogy with other boards, but builded image won't load

TheOnlyJoey, 2025-09-28:
I suspect this is related to https://github.com/armbian/linux-rockchip/issues/379#issuecomment-3303574616
Seems like the current DTS's for Radxa boards are not function on Armbian

bmx666, 2025-11-29:
I recently got access to Rockchip Android 14 kernel 6.1, it contains a lot of updates related to MIPI-CSI2, MIPI-DSI, VOP and etc.
I'm waiting an opportunity to get access to RK3588 board with MIPI-CSI cameras... to check these patches.