armbian/linux-rockchip #201: Trying to compile rk35xx 6.1 dtbos

si0ls, 2024-07-01:
Hi guys 👋 

I'm trying to use this kernel to compile Talos Linux for Orange Pi 5 (rk3588s).
Everything seems to work, thanks a lot for your amazing work 😍

But I still have a little problem, I dont success building the dtbos for the Rockchip platforms: it's not included with `make dtbs`.

Please could you help me achieving this?

Thanks a lot

igorpecovnik, 2024-07-01:
> to compile Talos Linux

We don't support 3rd party Linux distributions. Please consider using their support resources.

closed 2024-07-01 by igorpecovnik

si0ls, 2024-07-01:
I was just asking how to compile dtbos from the Makefile. I gave context for transparency but it's not really relevant there.

igorpecovnik, 2024-07-01:
> I was just asking

There are thousands of people "just asking" and seeking for attention. We don't have capacity to deal with that so we kindly asks you to use community forums.

This issue tracker is reserved for common (!) issues related to this kernel. This is not an issue of this kernel but support question -> https://forum.armbian.com/ If you use 3rd party Linux, its better to use their support resources as there might be certain specifics related to kernel deployments. Few people that provides you software for free can support all operating systems on extremely diverse hardware.

> thanks a lot for your amazing work

https://github.com/sponsors/armbian

amazingfate, 2024-07-01:
Arrmbian's build framework can build dtbos. So that's an issue of how you compile the kernel. We know nothing about that. You have to fix it by yourself.
Take a look at `arch/arm64/boot/dts/rockchip/overlay/Makefile`.