armbian/linux-rockchip #204: Suggestion to Optimize clk_put Usage in `rk628_csi_probe_of`

msmarks, 2024-07-06:
In the rk628_csi_probe_of function, the clk_put label is currently used to disable and unprepare the soc_24M clock regardless of whether an error occurs earlier in the function. However, it seems that this might not be necessary and could be optimized. Specifically, the clk_put section may only need to be executed when an error occurs, rather than on every exit of the function.

https://github.com/armbian/linux-rockchip/blame/75ac2784c40818adbdabb06e98ac18317af5fdac/drivers/media/i2c/rk628/rk628_csi_v4l2.c#L2163

amazingfate, 2024-07-07:
Please check if rkr3 has fixed it.

amazingfate, 2024-07-07:
https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr3/drivers%2Fmedia%2Fi2c%2Frk628%2Frk628_csi_v4l2.c

msmarks, 2024-07-07:
It was fixed on branch rkr3, thanks.

closed 2024-07-07 by msmarks