armbian/linux-rockchip #219: Headphones not working at Nano PC T6

J-Twin, 2024-08-11:
Hello,

My headphones are not working using the latest versions. Here is what I get:

dmesg:
asoc-simple-card rt5616-sound: ASoC: DAPM unknown pin Headphones
input: realtek,rt5616-codec Headphones as /devices/platform/rt5616-sound/sound/card1/input2

In rk3588-nanopc-t6.dts I can see:
	rt5616_sound: rt5616-sound {
		status = "okay";
		compatible = "simple-audio-card";
		pinctrl-names = "default";
		pinctrl-0 = <&hp_det>;

		simple-audio-card,name = "realtek,rt5616-codec";
		simple-audio-card,format = "i2s";
		simple-audio-card,mclk-fs = <256>;

		simple-audio-card,hp-det-gpio = <&gpio1 RK_PC4 GPIO_ACTIVE_LOW>;
		simple-audio-card,hp-pin-name = "Headphone Jack";

		simple-audio-card,widgets =
			"Headphone", "Headphone Jack",     <<<<<<<<<<<<<<<<

So Headphone or Headphone Jack is expected but Headphones is used according to the error in dmesg.

The error is not in the dts file. It is at another place. Similar problem was seen for another board and codec combo:
https://patchwork.kernel.org/project/alsa-devel/patch/1462435421-21205-6-git-send-email-enric.balletbo@collabora.com/

Since I am not an Linux audio subsystem expert I am not sure where exactly the error is. A possible suspect is the following file:
/sound/soc/rockchip/rk3399_gru_sound.c
static struct snd_soc_jack_pin rockchip_sound_jack_pins[] = {
	{
		.pin = "Headphones",
		.mask = SND_JACK_HEADPHONE,
	},

My T6 device(non LTS) uses the RT5616 codec but I have not located any pin entry at the codec driver(sound/soc/codecs/rt5616.c) which is used along with simple-audio-card(/sound/soc/generic/simple-card.c)

Please have a look at it, I will be happy to test any patch.

Best regards,
Ioannis

J-Twin, 2024-08-11:
Hello,

Not to forget, friendlyelec kernel is not having this problem with 6.1.57 that I tested some time ago:
input: realtek,rt5616-codec Headphone Jack as /devices/platform/rt5616-sound/sound/card1/input3

Also this is the Armbian that is running at my board:
uname -a
Linux nanopct6 6.1.75-vendor-rk35xx #1 SMP Thu Aug  8 17:42:28 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Armbian 24.8.0-trunk.544 noble
Release:	24.04
Codename:	noble

Best regards,
Ioannis

ginkage, 2024-08-11:
FWIW, with our device, we had to apply the following patch for headphone jack: https://github.com/StonedEdge/Retro-Lite-CM5/blob/dce1e69ceb5d21e9f7d6073033ad776d6da86b99/armbian/userpatches/kernel/rk35xx-vendor-6.1/0001-Fix-headphone-jack.patch

J-Twin, 2024-08-11:
@ginkage Thanks for the reply. Wonder if this change will break any other device using Headphones entry

ginkage, 2024-08-11:
It might. 
For that reason, my current plan is to make the that
`simple-audio-card,hp-pin-name = "Headphone Jack";`
line actually do something (it's kind of ignored at the moment).
I'll get to that once I'm back from my vacation.

defencedog, 2024-09-15:
I have a [similar issue](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1041) with OPi3b using `rk817` codec only on vendor kernel. Everything works great on mainline. Perhaps people here can give me pointers as well.

SuperKali, 2024-10-02:
This error has been fixed on #253, I tried personally and everything's works correctly

closed 2024-10-02 by SuperKali