armbian/linux-rockchip #226: HDMI audio not working on Rock 5 ITX v1.11

MikeFarrington, 2024-08-21:
I am unable to get HDMI output audio to work on any [Armbian builds for this board](https://www.armbian.com/radxa-rock-5-itx/) (including nightly), whereas it works on the [Radxa-supplied Debian b6 build](https://github.com/radxa-build/rock-5-itx/releases/tag/b6) from June (but overall video playback performance of that distro is rather lacking).

My knowledge or Linux is rather superficial as I have only toyed with it occasionally over the years and have never daily driven it. But I'm happy to test out any builds and perform guided troubleshooting.

MikeFarrington, 2024-08-21:
Update: after playing around with other audio devices, I started getting HDMI audio.  I plugged in a USB wifi/bluetooth dongle and paired up a bluetooth headset.  To my surprise, I started getting audio out via HDMI (regardless of which audio output device was selected -- I couldn't actually get the audio to come out via Bluetooth).

SuperKali, 2024-09-11:
> HDMI port labeled "HDMI TX1 8k".

Actually on DTS is only present HDMI sound from 8k port: https://github.com/armbian/linux-rockchip/blob/71039841e6100b280bc90d91b7b6b9d33ec84897/arch/arm64/boot/dts/rockchip/rk3588-rock-5-itx.dts#L111

Should be integrated too for hdmi0.

For this reason, the primary port not give any sound.

SuperKali, 2026-05-15:
Hey @MikeFarrington, dropping by almost a year later — still happening for you on current Armbian builds?

If yes, could you boot with a display plugged into the TX2 4K port and paste:

```bash
# kernel side - what got bound?
dmesg | grep -iE "dp0|dp1|hdmi|dw-dp|usbdp|drm"

# ALSA side - which cards exist?
aplay -l
cat /proc/asound/cards

# from the card you'd expect to be "rockchip-hdmi0"
speaker-test -c 2 -D plughw:CARD=rockchiphdmi0,DEV=0 -t sine -f 440 -l 1
```

Short story: the TX2 4K port goes through an RA620 active DP→HDMI converter (Radxa-proprietary chip, no public datasheet). Upstream has a `radxa,ra620` simple-bridge driver in review on LKML (v7 by Andy Yan, Aug 2025) — until that lands the audio path on that connector is uncharted.

If it's working now / you've moved off the board, no worries, I'll close it out.