armbian/linux-rockchip #229: Wayland Cursor Invisible [drm:vop2_plane_atomic_check] *ERROR*

defencedog, 2024-08-27:
I have share detailed information on [armbian forums](https://forum.armbian.com/topic/44197-build-framework-patches-placement-for-orangepi-3b-v21/?do=findComment&comment=200197) but here is a concise stuff

My SBC is Opi3b v2.1 on `jammy` with `panfrost` enabled
Whether I install `ubuntu-desktop` or `armbian-gnome`; the cursor is invisible, however cursor works OK while logged in as `X-session` of Ubuntu meaning `wayland` is to blame. If you logged in KODI wayland session cursor works OK?? GDM3 has no cursor as well

When I move cursor while in wayland session, these errors keep repeating `journalctl -f`

```
Aug 26 20:41:02 orangepi3b gnome-shell[2921]: Failed to post KMS update: drmModeAtomicCommit: Invalid argument
Aug 26 20:41:02 orangepi3b kernel: [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
Aug 26 20:41:02 orangepi3b gnome-shell[2921]: Page flip discarded: drmModeAtomicCommit: Invalid argument
Aug 26 20:41:02 orangepi3b gnome-shell[2921]: Failed to post KMS update: drmModeAtomicCommit: Invalid argument
Aug 26 20:41:02 orangepi3b kernel: [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
Aug 26 20:41:02 orangepi3b gnome-shell[2921]: Page flip discarded: drmModeAtomicCommit: Invalid argument
Aug 26 20:41:02 orangepi3b gnome-shell[2921]: Failed to post KMS update: drmModeAtomicCommit: Invalid argument
Aug 26 20:41:02 orangepi3b kernel: [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
Aug 26 20:41:02 orangepi3b gnome-shell[2921]: Page flip discarded: drmModeAtomicCommit: Invalid argument
 
```
I strongly believe its some GNOME related error. Internet is full of this bug when people are using eGPUs. If anyone has any idea, please help

defencedog, 2024-08-27:
Similar error in @amazingfate [comment here](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/89#issuecomment-1681722997)

amazingfate, 2024-08-27:
I don't have the board to test. If you think it's an issue of gnome, you can try noble.

PR #234 (2024-09-11): arm64: dts: rockchip: Add Orange Pi 3B v2.1
### Changes
- Add dts for Orange Pi 3B v2.1 ([dts source](https://github.com/orangepi-xunlong/linux-orangepi/blob/38dd4d9d9a1578f173912529bd587f63d7bd81e0/arch/arm64/boot/dts/rockchip/rk3566-orangepi-3b-v2.dts))
- Rename dtb filenames for v1.1 and v2.1 for compatibility with mainline u-boot. Now we can rely on u-boot to automatically select the appropriate dtb. ([u-boot reference](https://github.com/u-boot/u-boot/commit/a52099b4a2ae9e8cafc79268325249bcad308012#diff-1ca8d9ce95828f53150c73714c349fd38c171c4c3d51c2e606fea59a4358af84R13-R14))
- Keep old d [... cut]
closed 2024-09-11 by amazingfate