armbian/linux-rockchip #246: IMX219  get_mbus_config fail

smzahraee, 2024-09-22:
On connecting IMX219, I got the error ` imx219 2-0010 get_mbus_config fail, pls check it.` for kernel 5.10.160 it is okay but for kernel 6.1 this problem exists

```bash
[    6.509422] rockchip-iodomain fdc20000.syscon:io-domains: pmuio2(3300000 uV) supplied by vcc_3v3
[    6.509543] rockchip-iodomain fdc20000.syscon:io-domains: vccio1(3300000 uV) supplied by vccio_acodec
[    6.509634] rockchip-iodomain fdc20000.syscon:io-domains: vccio2(1800000 uV) supplied by vcc_1v8
[    6.509734] rockchip-iodomain fdc20000.syscon:io-domains: vccio3(3300000 uV) supplied by vccio_sd
[    6.509783] rockchip-iodomain fdc20000.syscon:io-domains: vccio4(1800000 uV) supplied by vcc_1v8
[    6.509830] rockchip-iodomain fdc20000.syscon:io-domains: vccio5(3300000 uV) supplied by vcc_3v3
[    6.509879] rockchip-iodomain fdc20000.syscon:io-domains: vccio6(3300000 uV) supplied by vcc_3v3
[    6.509925] rockchip-iodomain fdc20000.syscon:io-domains: vccio7(3300000 uV) supplied by vcc_3v3
[    6.510331] imx219 2-0010: Consider updating driver imx219 to match on endpoints
[    6.510381] rockchip-csi2-dphy csi2-dphy1: dphy1 matches imx219 2-0010:bus type 5
[    6.510401] rockchip-csi2-dphy csi2-dphy1: imx219 2-0010 get_mbus_config fail, pls check it
[    6.511024] rockchip-csi2-dphy csi2-dphy1: imx219 2-0010 get_mbus_config fail, pls check it
[    6.511057] rockchip-csi2-dphy csi2-dphy1: imx219 2-0010 get_mbus_config fail, pls check it
[    6.511084] rkisp-vir0: Async subdev notifier completed
[    6.511098] rockchip-csi2-dphy csi2-dphy1: csi2 dphy1 probe successfully!
[    6.511480] mpp_rkvenc fdf40000.rkvenc: Adding to iommu group 4

```

bmx666, 2024-12-03:
I found a patch for imx219 and ov5647 in another fork 

https://github.com/Joshua-Riek/linux-rockchip/commit/1b4e0edcaad78ed2b298318f8482f14838ac993b

igorpecovnik, 2024-12-03:
> I found a patch for imx219 and ov5647 in another fork

Grab it and provide a PR.

bmx666, 2024-12-03:
I got confirmation from customer on custom board based on rk3568, patched ov5647 can be captured and streaming.

<details>
<summary>Before patch for OV5647</summary>

root@smarc:~# dmesg | grep -i -e ov5647 -e csi -e mipi -e dphy
[    8.226269] platform csi2-dphy1: Fixed dependency cycle(s) with /rkisp-vir0/port/endpoint@0
[    8.805189] SCSI subsystem initialized
[    9.666901] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 241)
[    9.670025] rockchip-csi2-dphy-hw fe870000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[   10.197453] i2c 4-0036: Fixed dependency cycle(s) with /csi2-dphy1/ports/port@0/endpoint@1
[   10.236802] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: enter mipi csi2 hw probe!
[   10.237172] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: probe success, v4l2_dev:mipi-csi2-hw!
[   10.242327] rkisp rkisp-vir0: Entity type for entity rkisp-csi-subdev was not initialized!
[   10.337266] ov5647 4-0036: Consider updating driver ov5647 to match on endpoints
[   10.337298] rockchip-csi2-dphy csi2-dphy1: dphy1 matches ov5647 4-0036:bus type 5
[   10.337313] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   10.337775] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   10.337794] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   10.337822] rockchip-csi2-dphy csi2-dphy1: csi2 dphy1 probe successfully!
[   14.642550] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   14.655974] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   14.658456] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   14.667623] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   14.673766] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   14.675991] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   14.693690] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
[   14.700768] rockchip-csi2-dphy csi2-dphy1: ov5647 4-0036 get_mbus_config fail, pls check it
</details>

<details>
<summary>After patch for OV5647</summary>

root@smarc:~# dmesg | grep -i -e ov5647 -e csi -e mipi -e dphy
[    8.216025] platform csi2-dphy1: Fixed dependency cycle(s) with /rkisp-vir0/port/endpoint@0
[    8.795314] SCSI subsystem initialized
[    9.655310] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 241)
[    9.658522] rockchip-csi2-dphy-hw fe870000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[   10.184039] i2c 4-0036: Fixed dependency cycle(s) with /csi2-dphy1/ports/port@0/endpoint@1
[   10.187504] ov5647 4-0036: driver version: 00.01.01
[   10.187800] ov5647 4-0036: OV5647 power on
[   10.716071] ov5647 4-0036: OV5647 power off
[   10.718561] ov5647 4-0036: OmniVision OV5647 camera driver probed
[   10.722982] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: enter mipi csi2 hw probe!
[   10.723355] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: probe success, v4l2_dev:mipi-csi2-hw!
[   10.728530] rkisp rkisp-vir0: Entity type for entity rkisp-csi-subdev was not initialized!
[   10.819350] ov5647 4-0036: Consider updating driver ov5647 to match on endpoints
[   10.819386] rockchip-csi2-dphy csi2-dphy1: dphy1 matches m00_b_ov5647 4-0036:bus type 5
[   10.819940] rockchip-csi2-dphy csi2-dphy1: csi2 dphy1 probe successfully!
</details>

Waiting feedback for patched IMX219 driver.

igorpecovnik, 2024-12-03:
> But rk3588 has some issues with kernel 6.1

"Some"? :) Well, this https://github.com/armbian/linux-rockchip/pull/289 is best what is publicly available and we have negative support budget to maintain this stack.

bmx666, 2024-12-03:
@igorpecovnik relax man, no one blamed or shamed you. I jumped into Rockchip 2 months ago, and deal with customized USB, PCIe, DisplayPort, HDMI and I2S for different codecs for the customers. Jumpred into MIPI-CSI 3-4 weeks ago, and I'm trying to support customer into this area. Luck of documentation from Rockchip makes it's even harder. Binding these cameras via DTS another story.

Meanwhile from the latest customer update about patched IMX219, it works on rk3568 also.
Patched imx219 can be captured and streaming.

<details>
<summary>Before patch for IMX219</summary>

root@smarc:~# dmesg | grep -i -e imx219 -e csi -e mipi
[    8.222906] platform csi2-dphy1: Fixed dependency cycle(s) with /rkisp-vir0/port/endpoint@0
[    8.801019] SCSI subsystem initialized
[    9.664466] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 241)
[    9.667560] rockchip-csi2-dphy-hw fe870000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[   10.190952] i2c 4-0010: Fixed dependency cycle(s) with /csi2-dphy1/ports/port@0/endpoint@1
[   10.194076] imx219 4-0010: Looking up VANA-supply from device tree
[   10.194111] imx219 4-0010: Looking up VANA-supply property in node /i2c@fe5d0000/cam0_imx219@10 failed
[   10.194172] imx219 4-0010: supply VANA not found, using dummy regulator
[   10.194388] imx219 4-0010: Looking up VDIG-supply from device tree
[   10.194414] imx219 4-0010: Looking up VDIG-supply property in node /i2c@fe5d0000/cam0_imx219@10 failed
[   10.194462] imx219 4-0010: supply VDIG not found, using dummy regulator
[   10.194561] imx219 4-0010: Looking up VDDL-supply from device tree
[   10.194587] imx219 4-0010: Looking up VDDL-supply property in node /i2c@fe5d0000/cam0_imx219@10 failed
[   10.194629] imx219 4-0010: supply VDDL not found, using dummy regulator
[   10.208752] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: enter mipi csi2 hw probe!
[   10.209134] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: probe success, v4l2_dev:mipi-csi2-hw!
[   10.214278] rkisp rkisp-vir0: Entity type for entity rkisp-csi-subdev was not initialized!
[   10.303838] imx219 4-0010: Consider updating driver imx219 to match on endpoints
[   10.303877] rockchip-csi2-dphy csi2-dphy1: dphy1 matches imx219 4-0010:bus type 5
[   10.303891] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   10.304357] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   10.304378] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   10.304405] rockchip-csi2-dphy csi2-dphy1: csi2 dphy1 probe successfully!
[   14.955735] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   14.966300] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   14.972171] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   14.981961] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   14.988296] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   15.010605] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   87.604836] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   87.669250] rockchip-csi2-dphy csi2-dphy1: imx219 4-0010 get_mbus_config fail, pls check it
[   87.676575] rockchip-csi2-dphy1: dphy1, data_rate_mbps 912
</details>

<details>
<summary>After patch for IMX219</summary>

root@smarc:~# dmesg | grep -i -e imx219 -e csi -e mipi -e dphy
[    8.218330] platform csi2-dphy2: Fixed dependency cycle(s) with /rkisp-vir1/port/endpoint@0
[    8.797975] SCSI subsystem initialized
[    9.659508] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 241)
[    9.662701] rockchip-csi2-dphy-hw fe870000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[   10.188661] i2c 3-0010: Fixed dependency cycle(s) with /csi2-dphy2/ports/port@0/endpoint@1
[   10.192448] imx219 3-0010: driver version: 00.01.01
[   10.217353] imx219 3-0010: Model ID 0x0219, Lot ID 0x818c83, Chip ID 0x00f2
[   10.217397] ==> imx219 ken: 1104 imx219_probe
[   10.221868] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: enter mipi csi2 hw probe!
[   10.222252] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: probe success, v4l2_dev:mipi-csi2-hw!
[   10.227458] rkisp rkisp-vir1: Entity type for entity rkisp-csi-subdev was not initialized!
[   10.320906] imx219 3-0010: Consider updating driver imx219 to match on endpoints
[   10.320943] rockchip-csi2-dphy csi2-dphy2: dphy2 matches m01_b_imx219 3-0010:bus type 5
[   10.321692] rockchip-csi2-dphy csi2-dphy2: csi2 dphy2 probe successfully!
</details>

bmx666, 2024-12-03:
P.S. the customer kernel was based on branch rk-6.1-rkr3

igorpecovnik, 2024-12-03:
> relax man, no one blamed or shamed you

Didn't take it that way, no worries. Heard that too many times. 

> 1 week of pain

Understand. Aim of Armbian is to address and mitigate this common pain. With what we have, as much as possible, best effort.

> Luck of documentation from Rockchip makes it even harder.

This is what they sell, alongside with support. In case you want to speed up and your company can cover, I can privately [link](https://www.armbian.com/contact/) you with a person that knows more and is operating in this area. Just an idea.

> Patched imx219 can be captured and streaming.

Great to hear that!

belveder79, 2025-03-21:
@bmx666 these things are all related to a bug noone was able to find since mid of 2024... the 6.1 kernel is flawed, relating also to this post
https://forum.radxa.com/t/dev-video-01/20490/34
and to another issue report here:
https://github.com/armbian/linux-rockchip/issues/164

bmx666, 2025-03-21:
@belveder79 at least we have good news: Kernel 5.10 EOL is Dec 2026, and Android will support kernel security patches until July 1, 2027. Android 14 is not an option with kernels 5.15 and 6.1 and Android 15 requires a minimum kernel version of 6.6, which might push Rockchip to migrate to a newer kernel.

bmx666, 2025-09-27:
> just to say that the current armbian branch rk-6.1-rkr5 is still broken...

I will get soon access to new Android 15 for RK3588 and RK3588s, hope it will contain kernel 6.1+ and not the old kernel 5.10. In that case I could analyze and PR some patches into ISP/CIF/MIPI-CSI.

bmx666, 2025-11-30:
> [@bmx666](https://github.com/bmx666) Yes we were discussing this on multiple forums... I don't have the time to walk through the legacy friendlyelec kernel and patch the armbian kernel accordingly. For now I'm working with the official friendlyelec kernel which I compiled myself. That one does not support overlays though afaik, so custom compiling is a must. I've got all my cameras to work with the friendlyelec version without any pain, but for the armbian kernel it just seems that someone needs to look into this issue and fix it some time...

Do you mean this Android kernel -> https://github.com/friendlyarm/kernel-rockchip ?

saeedvft, 2026-02-07:
I’m experiencing the same get_mbus_config fail issue with an IMX219 camera on an RK3576-based board (AIO-3576Q with TF-ICORE-33576Q38-BTB and iCore-3576Q38 on top of that). The camera is connected via i2c6 and is detected, but I’m seeing continuous errors in the kernel log:
```
root@rk3576-buildroot:/# dmesg | grep imx
[    3.727293] platform csi2-dphy0: Fixed dependency cycle(s) with /i2c@2ac90000/imx219@10
[    3.835824] imx219 6-0010: Consider updating driver imx219 to match on endpoints
[    3.835840] rockchip-csi2-dphy csi2-dphy0: dphy0 matches imx219 6-0010:bus type 5
[    3.835854] rockchip-csi2-dphy csi2-dphy0: imx219 6-0010 get_mbus_config fail, pls check it
[    4.290246] rkcif-mipi-lvds1: rkcif_update_sensor_info: get terminal imx219 6-0010 g_frame_interval failed!
```

This prevents the camera from being usable. The issue occurs with kernel 6.1, but I understand from the discussion that similar problems exist on RK3568/RK3588 and were resolved with patches to the IMX219 driver.

Could the patches from PR #313 or #316 be adapted for RK3576? If not, are there any known workarounds or updates for this platform?

### Additional context:

    Using the RK3576 SDK.

    The camera is recognized but fails during configuration.

    No video device (/dev/video0) is created for capture.

Any guidance or updates on resolving this would be greatly appreciated.

saeedvft, 2026-02-09:
> According to the latest update, part of CSI is still under development, but `PHY MIPI CSI DPHY (inno)` was completed in 6.18. And the `MIPI CSI host` will be implemented in kernel 6.20 - https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md
> But `ISP` is still in `TODO`, so it is not yet known when it will be available.

So what about IMX415 module?
I checked the IMX415 driver in the sdk and it had implementation for  `get_mbus_config` function. Also you can see that rockchip applied it's own patch on IMX415; But the IMX219 driver is completely irrelevant:

IMX219:
```
// SPDX-License-Identifier: GPL-2.0
/*
 * A V4L2 driver for Sony IMX219 cameras.
 * Copyright (C) 2019, Raspberry Pi (Trading) Ltd
 *
 * Based on Sony imx258 camera driver
 * Copyright (C) 2018 Intel Corporation
 *
 * DT / fwnode changes, and regulator / GPIO control taken from imx214 driver
 * Copyright 2018 Qtechnology A/S
 *
 * Flip handling taken from the Sony IMX319 driver.
 * Copyright (C) 2018 Intel Corporation
 *
 */
```
IMX415:
```
// SPDX-License-Identifier: GPL-2.0
/*
 * imx415 driver
 *
 * Copyright (C) 2020 Rockchip Electronics Co., Ltd.
 *
 * V0.0X01.0X00 first version.
 * V0.0X01.0X01
 *  1. fix hdr ae ratio error,
 *     0x3260 should be set 0x01 in normal mode,
 *     should be 0x00 in hdr mode.
 *  2. rhs1 should be 4n+1 when set hdr ae.
 * V0.0X01.0X02
 *  1. shr0 should be greater than (rsh1 + 9).
 *  2. rhs1 should be ceil to 4n + 1.
 * V0.0X01.0X03
 *  1. support 12bit HDR DOL3
 *  2. support HDR/Linear quick switch
 * V0.0X01.0X04
 * 1. support enum format info by aiq
 * V0.0X01.0X05
 * 1. fixed 10bit hdr2/hdr3 frame rate issue
 * V0.0X01.0X06
 * 1. support DOL3 10bit 20fps 1485Mbps
 * 2. fixed linkfreq error
 * V0.0X01.0X07
 * 1. fix set_fmt & ioctl get mode unmatched issue.
 * 2. need to set default vblank when change format.
 * 3. enum all supported mode mbus_code, not just cur_mode.
 * V0.0X01.0X08
 * 1. add dcphy param for hdrx2 mode.
 */
```
So did you test IMX415 module?
Or did you try adopting the IMX219 driver based on IMX415?

bmx666, 2026-02-09:
> So what about IMX415 module? I checked the IMX415 driver in the sdk and it had implementation for `get_mbus_config` function. Also you can see that rockchip applied it's own patch on IMX415; But the IMX219 driver is completely irrelevant:
> 
> So did you test IMX415 module? Or did you try adopting the IMX219 driver based on IMX415?

Did you even read the history? Why some people can't scroll up and read a few comments?
And if you still lazy it's here -> https://github.com/armbian/linux-rockchip/issues/246#issuecomment-2513392342