armbian/linux-rockchip #266: Add the new RKNPU kernel version 0.9.8

vincenzodentamaro, 2024-10-17:
### Feature Description

As per the title, please add the new RKNPU kernel version 0.9.8 contained here https://github.com/airockchip/rknn-llm/tree/main/rknpu-driver

In the documentation, rockchip says that if the build fails modify the kernel header file kernel/include/linux/mm.h by adding the following code: 
```

static inline void vm_flags_set(struct vm_area_struct *vma, vm_flags_t flags){ 
vma->vm_flags |= flags; 
} 
static inline void vm_flags_clear(struct vm_area_struct *vma, vm_flags_t flags){ 
vma->vm_flags &= ~flags; 
} 
```

please update the releases with such kernel driver ASAP. If possible compile it as external module to be loaded with modprobe so if they update the driver, we can recompile it and load it with modprobe instead of recompiling the kernel.

Thank you

igorpecovnik, 2024-10-18:
https://github.com/armbian/linux-rockchip/pulls

amazingfate, 2025-01-02:
We have toggled to rkr4.1 with rknpu 0.9.8.

closed 2025-01-02 by amazingfate