armbian/linux-rockchip #279: [Feature Request]: Support for Orange Pi 5 Max in Latest 6.1x Linux Kernel

shenmintao, 2024-11-04:
### Which feature would you like to have?

Hello,

I came across a new device tree (dts) created for the Orange Pi 5 Max in the following link:
https://lore.kernel.org/linux-rockchip/20241031031528.23428-4-honyuenkwun@gmail.com/

It appears to be compatible with the 6.1x Linux kernel. Could we consider merging this patch to enable support for the Orange Pi 5 Max in the latest kernel version? Additionally, are there any further steps we need to take to ensure that the Orange Pi 5 Max can effectively utilize the new kernel?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2024-11-04:
That issue move makes no sense since this is unrelated to `vendor` kernel. 6.1x != 6.1.x

igorpecovnik, 2024-11-04:
Ups. But I think he wanted vendor while pooling from mainline. In any case, we will do nothing.