armbian/linux-rockchip #288: rk3528 kernel  WARNING： vop2  |  rk_gmac-dwmac

eternal-Eason, 2024-11-12:
os：Armbian-unofficial 24.11.0-trunk noble
kernel：Linux version 6.1.75  rk3

Hello, I encounter warning messages when starting the kernel on my rk3528 board running Armbian.

HDMI and GMAC are functioning normally.

The DTS file I'm using is rk3528-demo4-ddr4-v10-linux.dts, which is provided by Rockchip.

Switching to other DTS files also results in the same error messages.

Kernel commit: c0cb0d03161322215bb4d21740a7260d62f3c4c2.

eternal-Eason, 2024-11-12:
[kernel.log](https://github.com/user-attachments/files/17713788/kernel.log)

amazingfate, 2024-11-13:
The kernel error arre happening at early driver probe stage(at initramfs stage). 
Maybe we can try building vop2 driver and gpiolib driver as module:
```
CONFIG_ROCKCHIP_VOP2=m
CONFIG_GPIOLIB=m
```

eternal-Eason, 2024-11-13:
谢谢，解决了，不是什么大问题，是芯片兼容性问题。

`
+&gmac0 {
+       status = "okay";
+       snps,reset-gpio = <&gpio4 RK_PD7 GPIO_ACTIVE_LOW>;
+};
+
 &tve_in_vp1 {
        status = "disabled";
 };
diff --git a/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c b/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c
index ad2952318fb8..219ae191e25e 100644
--- a/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c
+++ b/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c
@@ -13759,8 +13759,8 @@ static char *vop2_plane_mask_to_string(unsigned long mask)
 
 static inline const char *vop2_plane_id_to_string(unsigned long phy)
 {
-       if (WARN_ON(phy >= ARRAY_SIZE(vop2_layer_name_list)))
-               return NULL;
+       //if (WARN_ON(phy >= ARRAY_SIZE(vop2_layer_name_list)))
+       //      return NULL;
 
        return vop2_layer_name_list[phy];
 }
`

closed 2024-11-13 by eternal-Eason