armbian/linux-rockchip #297: Kernel panic with any RK3568 RKNPU

bmx666, 2024-11-25:
<details>
<summary>Kernel panic dump</summary>

```
[    9.611892] random: crng init done
[    9.622828] rockchip-pm-domain fdd90000.power-management:power-controller: failed to get ack on domain 'npu', target_idle = 0, target_ack = 0, val=0x1ee
[    9.622902] Kernel panic - not syncing: panic_on_set_idle set ...
[    9.624647] CPU: 2 PID: 1 Comm: swapper/0 Not tainted 6.1.75-vendor-rk35xx #1
[    9.625309] Hardware name: FriendlyElec NanoPi R5S C1 (DT)
[    9.625813] Call trace:
[    9.626056]  dump_backtrace+0xe4/0x108
[    9.626425]  show_stack+0x20/0x30
[    9.626747]  dump_stack_lvl+0x7c/0xa0
[    9.627107]  dump_stack+0x18/0x34
[    9.627427]  panic+0x150/0x328
[    9.627730]  rockchip_pmu_set_idle_request+0x10c/0x1cc
[    9.628222]  rockchip_pd_power+0x43c/0x494
[    9.628615]  rockchip_pd_power_on+0x24/0x38
[    9.629021]  _genpd_power_on+0x11c/0x150
[    9.629405]  genpd_power_on+0x5c/0x150
[    9.629762]  __genpd_dev_pm_attach+0x19c/0x26c
[    9.630193]  genpd_dev_pm_attach+0x68/0x6c
[    9.630586]  dev_pm_domain_attach+0x20/0x3c
[    9.630991]  platform_probe+0x58/0xc0
[    9.631351]  really_probe+0x1cc/0x390
[    9.631709]  __driver_probe_device+0x140/0x158
[    9.632137]  driver_probe_device+0x48/0xd0
[    9.632530]  __driver_attach+0xbc/0x17c
[    9.632900]  bus_for_each_dev+0x78/0xc4
[    9.633269]  driver_attach+0x2c/0x38
[    9.633614]  bus_add_driver+0xf8/0x20c
[    9.633971]  driver_register+0xbc/0xf8
[    9.634329]  __platform_driver_register+0x30/0x3c
[    9.634783]  rk_iommu_driver_init+0x24/0x30
[    9.635190]  do_one_initcall+0x94/0x1e4
[    9.635561]  kernel_init_freeable+0x2e8/0x2f0
[    9.635982]  kernel_init+0x30/0x138
[    9.636319]  ret_from_fork+0x10/0x20
[    9.636668] SMP: stopping secondary CPUs
[    9.637045] CPU1: stopping
[    9.637047] CPU3: stopping
[    9.637048] CPU0: stopping
[    9.637057] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 6.1.75-vendor-rk35xx #1
[    9.637070] Hardware name: FriendlyElec NanoPi R5S C1 (DT)
[    9.637075] Call trace:
[    9.637079]  dump_backtrace+0xe4/0x108
[    9.637096]  show_stack+0x20/0x30
[    9.637108]  dump_stack_lvl+0x7c/0xa0
[    9.637119]  dump_stack+0x18/0x34
[    9.637128]  local_cpu_stop+0x4c/0x6c
[    9.637139]  ipi_handler+0x188/0x1d0
[    9.637148]  handle_percpu_devid_irq+0x70/0x128
[    9.637162]  handle_irq_desc+0x28/0x40
[    9.637176]  generic_handle_domain_irq+0x24/0x30
[    9.637189]  gic_handle_irq+0x1ac/0x228
[    9.637198]  call_on_irq_stack+0x24/0x4c
[    9.637209]  do_interrupt_handler+0xa4/0xc4
[    9.637222]  el1_interrupt+0x94/0xa4
[    9.637233]  el1h_64_irq_handler+0x18/0x24
[    9.637245]  el1h_64_irq+0x74/0x78
[    9.637253]  arch_cpu_idle+0x18/0x44
[    9.637264]  default_idle_call+0x58/0x78
[    9.637273]  do_idle+0x128/0x250
[    9.637288]  cpu_startup_entry+0x3c/0x40
[    9.637302]  __cpu_disable+0x0/0xc0
[    9.637314]  __secondary_switched+0xb0/0xb4
[    9.637329] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 6.1.75-vendor-rk35xx #1
[    9.637341] Hardware name: FriendlyElec NanoPi R5S C1 (DT)
[    9.637347] Call trace:
[    9.637351]  dump_backtrace+0xe4/0x108
[... 169 lines cut ...]
[   10.755257]  27:          0          0          0          0     GICv3 168 Level     rockchip_usb2phy
[   10.756099]  31:          0          0          0          0     GICv3 107 Level     pcie-sys
[   10.756880]  32:          0          0          0          0     GICv3 104 Level     (null)
[   10.757649]  33:          0          0          0          0     GICv3  46 Level     fe530000.dma-controller
[   10.758550]  34:          0          0          0          0     GICv3  45 Level     fe530000.dma-controller
[   10.759438]  35:          0          0          0          0     GICv3 197 Level     pcie-sys
[   10.760221]  36:          0          0          0          0     GICv3  48 Level     fe550000.dma-controller
[   10.761119]  37:          0          0          0          0     GICv3  47 Level     fe550000.dma-controller
[   10.762019]  39:          0          0          0          0     GICv3 194 Level     (null)
[   10.762784] PMU:
[   10.762966] pwr       0x00a0: 0000007d 00000000 00000003 00000000
[   10.763543] status    0x0098: 0000007d 00000000 0000007d 00000000
[   10.764116] req       0x0050: 00000000 00000000 00000000 00000000
[   10.764687] idle      0x0068: 000001ee 00000000 0000ffff 0000000f
[   10.765259] ack       0x0060: 000001ee 00000000 000001ee 00000000
[   10.765831] Kernel Offset: 0x500bfce00000 from 0xffff800008000000
[   10.766392] PHYS_OFFSET: 0xffff82b340000000
[   10.766787] CPU features: 0xc0000,001400a4,6200721b
[   10.767243] Memory Limit: none
[   10.770972] ---[ end Kernel panic - not syncing: panic_on_set_idle set ... ]---
```

</details>

If disable rknpu in dts, then I could boot boards and reach console.

amazingfate, 2024-11-25:
This is not a kernel related issue, check regulator of npu in u-boot.

closed 2024-11-25 by amazingfate

bmx666, 2024-11-25:
Does this kernel 6.1 is compatible with this U-boot https://github.com/Kwiboo/u-boot-rockchip/tree/rk3xxx-2024.10 ?

amazingfate, 2024-11-26:
patch like this might help: https://github.com/armbian/build/blob/main/patch%2Fu-boot%2Fv2024.10%2Fboard_orangepi3b%2F0001-rockchip-rk3566-orangepi-3b-enable-npu-regulator.patch

bmx666, 2024-11-26:
@amazingfate  thanks a lot! You saved my day!