armbian/linux-rockchip #333: Unable to compile this project

ErezAlster, 2025-02-09:
I'm getting the following error during compilation:

> drivers/gpu/drm/panel/panel-innolux-afj101-ba2131.c:482:19: error: initialization of ‘void (*)(struct mipi_dsi_device *)’ from incompatible pointer type ‘int (*)(struct mipi_dsi_device *)’ [-Werror=incompatible-pointer-types]
  482 |         .remove = afj101_ba2131_dsi_remove,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/panel-innolux-afj101-ba2131.c:482:19: note: (near initialization for ‘afj101_ba2131_driver.remove’)
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: drivers/gpu/drm/panel/panel-innolux-afj101-ba2131.o] Error 1
make[4]: *** [scripts/Makefile.build:503: drivers/gpu/drm/panel] Error 2
make[3]: *** [scripts/Makefile.build:503: drivers/gpu/drm] Error 2
make[2]: *** [scripts/Makefile.build:503: drivers/gpu] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC      net/core/datagram.o
  CC      mm/vmstat.o
  CC      fs/nfs_common/nfs_ssc.o
  CC      kernel/sys.o

Is there documentation available about nightly builds available? is there cadence of releases available?

igorpecovnik, 2025-02-09:
> Is there documentation available about nightly builds available? 

Nightly builds are always compiling from trunk of https://github.com/armbian/build

Docs:
https://docs.armbian.com/Developer-Guide_Overview/

> is there cadence of releases available?

Nightly is just cron based build, nothing special. Point releases:
https://docs.armbian.com/Process_Release-Model/

amazingfate, 2025-02-10:
Werror is disable to get rid of these errors:
https://github.com/armbian/linux-rockchip/commit/43c9d5d831c5845d94639d95a29e63d08db96cdc

This kernel is from rockchip's SDK and we don't have much time to solve all the compile issues, so we use a quick method to fix.

closed 2025-03-14 by amazingfate