armbian/linux-rockchip #334: NPU Overlay missing for RK3566 device (Radxa Zero 3)

halabut69, 2025-02-11:
Linux Rockchip includes the RKNPU driver as standard for RK3566 builds. However, at least on a Radxa Zero 3W/E the NPU is not enabled without the correct Device Tree Overlay. This overlay is not included in the linux-rockchip overlays.
Without the overlay the npu is missing in /dev/ and there is no NPU driver loaded. Adding and building the overlay using armbian-add-overlay results in the NPU functioning correctly.
The overlay can be found here https://github.com/radxa-pkg/radxa-overlays/blob/main/arch/arm64/boot/dts/rockchip/overlays/rk3568-npu-enable.dts (this is probably forked from the Rockchip source, but I can't find it in their repository).

Suggest to add the required overlay to linux-rockchip

igorpecovnik, 2025-02-11:
> The overlay can be found here
> Suggest to add the required overlay to linux-rockchip

Download overlay, test it and prepare PR:
https://github.com/armbian/linux-rockchip/pulls

amazingfate, 2025-02-11:
I have no idea why radxa uses overlay to enable npu, on other board like opi3b npu is enabled by default.

halabut69, 2025-02-11:
Already tested, pull request added https://github.com/armbian/linux-rockchip/pull/335
@amazingfate some quick googling reveals that the Odroid M1S also needs this overlay.

igorpecovnik, 2025-02-11:
Odroid M1. There we only have mainline based kernel where this might not be supported yet.

rpardini, 2025-02-12:
Half the reason for running vendor kernel on 356x is the npu, I'd sincerely just enable it in the DT directly like the other boards. Overlay is fine but requires instructions, etc.

Odroid-m1, I've added a DT for it some time ago but it's probably not complete either.

halabut69, 2025-02-18:
Ok, conceptually no big issue to do this in the DT instead.
I'm no DT expert so one question:
Mostly it's just to add the reserved memory location, but there's a conflict with the existing zero3 .dtsi:
currently we have:
`&rknpu {
	rknpu-supply = <&vdd_gpu>;
};`

but the rockchip .dts has:
`&bus_npu {
	bus-supply = <&vdd_logic>;
	pvtm-supply = <&vdd_cpu>;
	status = "okay";
};`

So these are specifying different voltage-tables and voltage-temperature behaviours, right?
Is there any issue with substituting the rockchip supplied information?

amazingfate, 2025-03-14:
> Ok, conceptually no big issue to do this in the DT instead. I'm no DT expert so one question: Mostly it's just to add the reserved memory location, but there's a conflict with the existing zero3 [...]

Different boards may differ in hardware design, and board level dts can overwrite soc level dts. If it is confirmed working, it should be fine.

halabut69, 2025-03-18:
Ok, new pull request created to add NPU enabling settings into the Radxa Zero 3 device-tree
https://github.com/armbian/linux-rockchip/pull/343
Works fine on my 2gb radxa zero 3.