armbian/linux-rockchip #347: The version control conflict markers in dw_hdmi-rockchip.c on the rk-6.1-rkr5.1 branch?

Lemon1151, 2025-04-06:
drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c

if (!max_frl_rate || (tmdsclk < HDMI20_MAX_RATE && mode.clock < HDMI20_MAX_RATE)) {
<<<<<<< HEAD
		dev_dbg(hdmi->dev, "use tmds mode\n");
=======
		dev_info_once(hdmi->dev, "use tmds mode\n");
>>>>>>> bcac1f517909 (drm: rockchip: dw_hdmi: report tmds mode once)
		hdmi->link_cfg.frl_mode = false;
		return;
	}

amazingfate, 2025-04-06:
rkr5.1 branch is still wip, this should be an errror introduced by patch rebasing https://github.com/armbian/linux-rockchip/commit/af688c46ea061e6ebcb4e3a8ae455e063fca4f10.
@HeyMeco

HeyMeco, 2025-04-06:
Was on vacation, will continue with this in mind next week. Thanks for tagging

Lemon1151, 2025-04-06:
> 正在度假，下周将继续牢记这一点。感谢您的标记

rk3568-rock-3a.dts

[image]

Lemon1151, 2025-04-06:
> rkr5.1 branch is still wip, this should be an errror introduced by patch rebasing [af688c4](https://github.com/armbian/linux-rockchip/commit/af688c46ea061e6ebcb4e3a8ae455e063fca4f10). [@HeyMeco](https://github.com/HeyMeco)

Are there any differences between the rkr5.1 branch and the rkr5 branch other than the kernel version? Will all commits from the rkr5 branch be synchronized to the rkr5.1 branch?

amazingfate, 2025-04-07:
> Are there any differences between the rkr5.1 branch and the rkr5 branch other than the kernel version? Will all commits from the rkr5 branch be synchronized to the rkr5.1 branch?

rkr5.1 is the sdk version released by rockchip, and all commits in rkr5 branch should get rebased.

HeyMeco, 2025-04-15:
See https://github.com/armbian/linux-rockchip/issues/349