armbian/linux-rockchip #360: Minimum frequency is too high for RK3588

mahdibx, 2025-05-09:
The RK3588 supports frequencies as low as 408MHz, but in armbian its minimum is 1008MHz, which makes the cpu run hotter and draw more power even when idle.

```
$ cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1008000 1200000 1416000 1608000 1800000

$ cat /sys/devices/system/cpu/cpufreq/policy6/scaling_available_frequencies
1200000 1416000 1608000 1800000 2016000 2208000 2400000
```

Could you please add the missing frequencies below 1Ghz, so that the governor can use them on low power modes.

The actual missing frequencies:
For cores 0, 1, 2 ,3 (Cortex-A55) -> **408000** / **600000** / **816000** 
For cores 4, 5, 6, 7 (Cortex-A76) -> **408000** / **600000** / **816000** / **1008000** 

Thank you!

EvilOlaf, 2025-05-20:
duplicate
https://forum.armbian.com/topic/52017-available-frequencies-for-rk3588-are-wrong/

mahdibx, 2025-08-13:
This was solved by using the vendor kernel, I was on current.