armbian/linux-rockchip #364: Analog audio output stopped working after 6.1.75 on opi5

deskwizard, 2025-06-03:
Hi,

After a lot of image writing and kernel swapping,
I have come to the conclusion that something
went wrong between kernel 6.1.75 and 6.1.84.

No kernel after 6.1.75 have audio output on the 3.5mm jack.
HDMI audio works without issues on all of them.
The DE (KDE of GNOME) doesn't make a difference.
I can see the jack detection working, but nothing comes out either option.

Any information I can provide that could help?
I have some basic knowledge, but that one seems out of my league,
but I'm not afraid to figure out stuff and help.

I had a look at the device tree, and I didn't notice anything different between kernel versions,
besides addresses changing but that's throughout the file, all the same offset, so I doubt it is related.

Thanks in advance, that would really make my day/week/month/year
Deskwizard

cannister-of-sparrows, 2025-07-31:
I wonder if this is relevant?
https://github.com/armbian/linux-rockchip/commit/0f41390f2fc69a142dd1a4e57af1dc30a23b2538

Looks like it's being resolved here:
https://github.com/armbian/build/pull/7695

deskwizard, 2025-07-31:
Uhm, I don't know, but I sure hope so! I haven't noticed any issues with other audio outputs besides the analog out though.

Thanks for taking the time, I appreciate it a lot.
I'm kind of new to all this and as usual, I seem to hit all the corner-cases

JFLim1, 2025-08-24:
Same issue for Orange Pi 5 Plus.  Analog/Headphone audio out stopped working after 6.1.75-vendor-rk35xx.

vlw, 2026-08-15:
@JFLim1 same thing on OPi 5 Plus, and it's not the DTS.

Vendor 6.1 binds the jack as `rockchip-es8388` / ES8323. After 6.1.75 Rockchip added mute switches that boot **off**, so detect still works and HDMI is fine, but nothing comes out of the 3.5 mm.

The commit is 4389dbd8ec41 (Shunhua Lan, 20 Jun 2024) — *ASoC: es8323: Add a switch for dapm route to control "Speaker/Headphone Power" widgets*. It's not in 6.1.75, it is in 6.1.84 and later (including 6.1.115). `OUT1` is the jack, `OUT2` is the speaker. Their own images unmute this from userspace; we don't. The pl330 thing from earlier is unrelated.

On a running box:

```bash
amixer -c rockchipes8388 sset OUT1 on
amixer -c rockchipes8388 sset Headphone on
amixer -c rockchipes8388 sset 'hp switch' on
speaker-test -D plughw:rockchipes8388 -c 2 -t wav
```

If there's no `rockchipes8388`, check `cat /proc/asound/cards` and use that `-c` number.

To keep it after reboot:

```bash
sudo alsactl store
sudo systemctl enable --now alsa-restore.service
```

Proper fix is shipping that asound.state (or defaulting the switches on). Reverting that commit also brings back 6.1.75 behaviour.

deskwizard, 2026-08-16:
@vlw 

Confirmed working! :)

I didn't have to run the commands you mentioned to persist the settings, they we're still correct after reboot.
For future me and everyone else, the 'speaker-test' uses pink noise by default, that's normal. I was worried for a bit.

For current kernel (6.8.43), all I had to do is (For the output jack):
`amixer -c rockchipes8388 sset 'Output 1' 100%`

I can't thank you enough, you made my day! Hell, my month! :)

efectn, 2026-08-16:
> [@JFLim1](https://github.com/JFLim1) same thing on OPi 5 Plus, and it's not the DTS.
> 
> Vendor 6.1 binds the jack as `rockchip-es8388` / ES8323. After 6.1.75 Rockchip added mute switches that boot **off**, so detect still works and HDMI is fine, but nothing comes out of the 3.5 mm.
> 
> The commit is [4389dbd](https://github.com/armbian/linux-rockchip/commit/4389dbd8ec41bf41282b41aa0d83133f1dd4d264) (Shunhua Lan, 20 Jun 2024) — _ASoC: es8323: Add a switch for dapm route to control "Speaker/Headphone Power" widgets_. It's not in 6.1.75, it is in 6.1.84 and later (including 6.1.115). `OUT1` is the jack, `OUT2` is the speaker. Their own images unmute this from userspace; we don't. The pl330 thing from earlier is unrelated.
> 
> On a running box:
> 
> amixer -c rockchipes8388 sset OUT1 on
> amixer -c rockchipes8388 sset Headphone on
> amixer -c rockchipes8388 sset 'hp switch' on
> speaker-test -D plughw:rockchipes8388 -c 2 -t wav
> If there's no `rockchipes8388`, check `cat /proc/asound/cards` and use that `-c` number.
> 
> To keep it after reboot:
> 
> sudo alsactl store
> sudo systemctl enable --now alsa-restore.service
> Proper fix is shipping that asound.state (or defaulting the switches on). Reverting that commit also brings back 6.1.75 behaviour.

Yes, creating asound.state should be proper fix i think.